xmos/xcore_iot #144: RTOS Cifar10 in examples runs error

kenanking, 2021-06-22:
**System information**
 - OS: Ubuntu20.04
 - SDK version: XTC 15.0.6

**Describe the problem**
Error occurs when I run the examlpe of cifar10 in RTOS. I followed the constructions in the readme, but I still met the time stamp error.

**Output**
```
xrun --xscope bin/cifar10.xe
start scheduler on tile 1
start scheduler on tile 0
Logical Core 0 initializing as FreeRTOS Core 0
Logical Core 0 initializing as FreeRTOS Core 0
Logical Core 1 initializing as FreeRTOS Core 1
Logical Core 1 initializing as FreeRTOS Core 1
Logical Core 2 initializing as FreeRTOS Core 2
Logical Core 2 initializing as FreeRTOS Core 2
Logical Core 3 initializing as FreeRTOS Core 3
Logical Core 4 initializing as FreeRTOS Core 4
Timestamp error: last 10433 (tile 0x0), current 9995 (tile 0x0)
Logical Core 3 initializing as FreeRTOS Core 3
Logical Core 4 initializing as FreeRTOS Core 4
Initializing tile 1 core 1 on core 1
Initializing tile 1 core 2 on core 2
Initializing tile 1 core 3 on core 3
Initializing tile 0, core 2 on core 2
Initializing tile 1 core 4 on core 4
Initializing tile 1 core 0 on core 0
Timestamp error: last 14581 (tile 0x0), current 14247 (tile 0x0)
Initializing tile 0, core 3 on core 3
FreeRTOS Core 1 initialized
Initializing tile 0, core 4 on core 4
Initializing tile 0, core 0 on core 0
Timestamp error: last 16152 (tile 0x0), current 15696 (tile 0x0)
FreeRTOS Core 2 initialized
FreeRTOS Core 3 initialized
Timestamp error: last 16950 (tile 0x0), current 16733 (tile 0x0)
Initializing tile 0, core 1 on core 1
FreeRTOS Core 4 initialized
FreeRTOS Core 2 initialized
FreeRTOS Core 0 initialized
FreeRTOS Core 3 initialized
FreeRTOS Core 4 initialized
Timestamp error: last 18971 (tile 0x0), current 18629 (tile 0x0)
Startup task running from tile 1 on core 1
FreeRTOS Core 0 initialized
FreeRTOS Core 1 initialized
Startup task running from tile 0 on core 2
Starting model runner task
Starting QSPI flash driver
Starting filesystem
Wait for input tensor...
Running inference...
xrun: Program received signal ET_ILLEGAL_RESOURCE, Resource exception.
      [Switching to tile[1] core[4] (dual issue)]
      thread_group_add (__func=Die: DW_TAG_<unknown> (abbrev = 114, offset = 2235000)

        has children: FALSE
        attributes:
                DW_AT_type (DW_FORM_ref4) constant ref: 2221882 (adjusted)
      0x8a8b8<error reading variable>, __argument=0xb4c58, __stack_base=0x80080) at /home/tang/XMOS/XTC/15.0.6/target/include/xcore/thread.h:79
      79          __xcore_set_thread_worker(__xthread, __func);
      Current language:  auto; currently c++
```

keithm-xmos, 2021-06-22:
Hi @kenanking, we are sorry you are experiencing this issue.  A fix was just pushed to the `main` branch.  Also, the "Timestamp" errors are not true errors and can be ignored.  These will be fixed in a future release of our toolchain.

kenanking, 2021-06-23:
> Hi @kenanking, we are sorry you are experiencing this issue. A fix was just pushed to the `main` branch. Also, the "Timestamp" errors are not true errors and can be ignored. These will be fixed in a future release of our toolchain.

The problem has been solved, thank you!

closed 2021-06-23 by kenanking