xmos/xcore_iot #162: Update to C-friendly XUD release

keithm-xmos, 2021-07-30:
Interrupt handler versions of XUD_GetData(), XUD_GetSetupData(), and XUD_SetData() exist in lib_xud already as XUD_GetData_Select() and XUD_SetData_Select(), and their counterparts XUD_SetReady_Out() and XUD_SetReady_In(). However, the prototypes for these are only available for XC (surrounded by #if defined(__XC__)), and there is no equivalent for XUD_GetSetupData().  

As a temporary workaround, these were copied into the SDK and modified.  Ideally lib_xud will be updated to be fully compatible with standard C. 

Waiting on https://github.com/xmos/lib_xud/issues/128

keithm-xmos, 2022-03-02:
This is no longer blocked.

keithm-xmos, 2022-05-10:
No longer blocked as https://github.com/xmos/lib_xud/issues/349 was a false alarm.  However, an initial investigation leads us to believe that this is now a much bigger effort.

xmos-jmccarthy, 2022-05-10:
For future reference:
https://github.com/xmos/xcore_sdk/tree/feature/lib_xud_capi_update

This branch contains a lib_xud submodule updated to the current head of develop, updated sdk CMakeLists.txt, and trivial change for the removal of the function XUD_SetDevAddr().  This was found to build but was not functional.  Switching back to the old submodule, but with the new port layer, functioned as expected.