xmos/xcore_iot #295: Build of documentation failed

eesctc, 2022-02-25:
**System information**
 - OS: Ubuntu 20.04.3 LTS
 - SDK version tag or commit SHA: 0.9.5-a0

**Describe the problem**
Attempting to build the documentation as described in doc/readme.rst resulted in an error. I re-ran as su, but got another error.
It looks like the docker command is expecting the doc folder to be in a build folder with a settings.json file
It's possible that this is expecting other builds (cmake etc) to have have been run before, but this is not clear and the only information on building seems to be in the examples folder.  

**Provide the exact sequence of commands / steps that you executed before running into the problem**
git clone --recurse-submodules https://github.com/xmos/xcore_sdk.git
cd xcore_sdk/doc/
docker pull ghcr.io/xmos/doc_builder:main
sudo apt  install docker.io
docker run --rm -t -u "$(id -u):$(id -g)" -v $(pwd):/build -e REPO:/build -e DOXYGEN_INCLUDE=/build/doc/Doxyfile.inc -e EXCLUDE_PATTERNS=/build/doc/exclude_patterns.inc -e DOXYGEN_INPUT=ignore ghcr.io/xmos/doc_builder:main
sudo docker run --rm -t -u "$(id -u):$(id -g)" -v $(pwd):/build -e REPO:/build -e DOXYGEN_INCLUDE=/build/doc/Doxyfile.inc -e EXCLUDE_PATTERNS=/build/doc/exclude_patterns.inc -e DOXYGEN_INPUT=ignore ghcr.io/xmos/doc_builder:main

**Other info or logs** 
sudo docker run --rm -t -u "$(id -u):$(id -g)" -v $(pwd):/build -e REPO:/build -e DOXYGEN_INCLUDE=/build/doc/Doxyfile.inc -e EXCLUDE_PATTERNS=/build/doc/exclude_patterns.inc -e DOXYGEN_INPUT=ignore ghcr.io/xmos/doc_builder:main
Unable to find image 'ghcr.io/xmos/doc_builder:main' locally
main: Pulling from xmos/doc_builder
33847f680f63: Pull complete 
b693dfa28d38: Pull complete 
ef8f1a8cefd1: Pull complete 
248d7d56b4a7: Pull complete 
edfc8868abd9: Pull complete 
1ea9965ac650: Pull complete 
61cd5f9962c3: Pull complete 
6f8db6dbbc96: Pull complete 
34a68a957e28: Pull complete 
8fe2cbdf88be: Pull complete 
e75b361616fe: Pull complete 
14296f511732: Pull complete 
b582e0db7d13: Pull complete 
ce9c4764fcd5: Pull complete 
4f4fb700ef54: Pull complete 
332a2116db6c: Pull complete 
e732eb07f6eb: Pull complete 
b6839246a15a: Pull complete 
abbcbf26eb03: Pull complete 
088c3e407ad2: Pull complete 
31b76fae0d37: Pull complete 
f24db774745d: Pull complete 
9c7b03bcd736: Pull complete 
Digest: sha256:126b0ed7b89ba20cbf7ab9531bed87f128241eb95d6570a78dedf17b1c3e2461
Status: Downloaded newer image for ghcr.io/xmos/doc_builder:main
jq: error: Could not open file /build/settings.json: No such file or directory
jq: error: Could not open file /build/settings.json: No such file or directory
Windows users are recommended to run under wsl2 to make mounting and permissions easier and more performant
== XMOS docs is not initialized ==
 - Bind mount a repo with a "doc" folder to /build to work on existing documentation
 >    docker run -t -v $REPO_PATH:/build -u "$(id -u):$(id -g)" doc_builder

keithm-xmos, 2022-02-25:
Hello @eesctc,

Thank you trying the xcore_sdk and I apologize that you ran into an issue.  I do have a swift resolution for you.  The `docker run` command need to be run from the **root** of the repository.  It appears from your helpful information above that you ran from the `xcore_sdk/doc` directory when you really want just `xcore_sdk`.  Then run the command from the documentation README:

`$ docker run --rm -t -u "$(id -u):$(id -g)" -v $(pwd):/build -e REPO:/build -e DOXYGEN_INCLUDE=/build/doc/Doxyfile.inc -e EXCLUDE_PATTERNS=/build/doc/exclude_patterns.inc -e DOXYGEN_INPUT=ignore ghcr.io/xmos/doc_builder:main`

This should resolve your issue.  

Cheers,
@keithm-xmos

eesctc, 2022-02-28:
Yes, that worked. Thanks.

closed 2022-02-28 by eesctc