xmos/xcore_iot #435: Git submodules should be defined using https to permit cloning by non-authenticated users

robertbcalhoun, 2022-05-23:


IOSAudio, 2022-12-15:
Hi Guys,

I am hitting the incorrect rights on the submodules.

What is the correct way to clone this repository?

Many Thanks

Andy

AndrewCapon, 2022-12-15:
Ok, I worked out what is going wrong here.

First you need a GitHub account.

Then you need to add your default public ssh key into that account (.ssh/id_rsa.pub)

Then you can clone the "public" repository.

I had it set up with named ssh files (.ssh/id_rsa_<githhubaccount>.pub), this does not work.

keithm-xmos, 2022-12-20:
Hi @AndrewCapon, I apologize that you ran into this issue.  We do have plans to make modifications that will allow anonymous cloning.

keithm-xmos, 2023-02-17:
Won't fix.  Will revist in future if needed.

closed 2023-02-17 by keithm-xmos