xmos/xcore_iot #458: freertos explorer board example fails in run time

uvvpavel, 2022-05-31:
While running freertos explorer_board example on my windows machine, I got this 
`Failed to open file /flash/fs/demo.txt
xrun: Program received signal ET_LOAD_STORE, Memory access exception.
      [Switching to tile[0] core[4]]
      outs (lp=0x0) at /home/paveluvarov/sandboxes/xcore_sdk/modules/core/modules/utils/src/xcore_printf.c:92

      92        /home/paveluvarov/sandboxes/xcore_sdk/modules/core/modules/utils/src/xcore_printf.c: No such file or directory.
        in /home/paveluvarov/sandboxes/xcore_sdk/modules/core/modules/utils/src/xcore_printf.c`

keithm-xmos, 2022-05-31:
@uvvpavel, did you get any errors when you ran the step to flash the filesystem?

`make flash_fs_example_freertos_explorer_board`

or

`nmake flash_fs_example_freertos_explorer_board`

uvvpavel, 2022-06-01:
Closing this issue, since I've been using xflash wrong

closed 2022-06-01 by uvvpavel