xmos/xcore_iot #504: Build system example doesn't include the SDK

xhuw, 2022-08-11:
**Describe the documentation issue**

Just reading the SDK documentation, spotted [this page](https://github.com/xmos/xcore_sdk/blob/main/doc/tutorials/build_system/cmakelists.rst). 

Those CMakeLists examples never include the SDK, so it won't exist when the SDK is added. I would expect to see a call to `include`/`add_subdirectory`/`find_package`/`externalproject_add` so that the SDK libraries being linked in will exist.

**Other information**

xmos-jmccarthy, 2022-08-11:
In the past there has been much debate on whether people should develop their application inside or outside of the SDK.  Due to this we are intentionally not prescriptive inside this repository, because it is up to the application developer to decide if they want to put their application inside the xcore_sdk repository or outside of the xcore_sdk repository.  

The expected entry point for most users is at the sln_* level, not the xcore_sdk  repository itself, which should provide more clarity.  See https://github.com/xmos/sln_voice, in particular https://github.com/xmos/sln_voice/blob/develop/CMakeLists.txt

That all being said, I have updated the doc page referenced here to include that the repository is compatible with add_subdirectory().
PR #506

xhuw, 2022-08-12:
Thanks for the thorough response. Update to the page looks good. I agree it's good to let the user chose how to use the SDK.

PR #506 (2022-08-14): Improve cmake doc
Closes #504
closed 2022-10-18 by xmos-jmccarthy