xmos/xcore_iot #507: Building an app with xs3a.cmake gives a warning

xhuw, 2022-08-12:
**System information**
 - OS: WSL
 - xcore board 3600-EVK
 - SDK version tag or commit SHA 67bf6b5041c90a5be711124e3fa708a64d9a3c2c

**Describe the current behavior**

When using xs3a.cmake as the toolchain for my project prints

```
System is unknown to cmake, create:
Platform/XCORE_XS3A to use this system, please post your config file on discourse.cmake.org so it can be added to cmake
```

**Describe the expected behavior**

cmake should look happy when it does a build

**Standalone code to reproduce the issue**

In a directory with a CMakeLists.txt and a clone of the sdk run:
```
cmake -DCMAKE_TOOLCHAIN_FILE=xcore_sdk/xmos_cmake_toolchain/xs3a.cmake
```

**Other info or logs** 

@CiaranWoodward sent me some helpful links:
[1](https://gitlab.kitware.com/cmake/cmake/-/issues/12521#note_1020913) [2](https://gitlab.kitware.com/cmake/cmake/-/merge_requests/6429) [3](https://gitlab.kitware.com/cmake/cmake/-/blob/master/CONTRIBUTING.rst) [4](https://gitlab.kitware.com/DaveAtkinsonXMOS) [5](https://discourse.cmake.org/t/contribution-cross-compilation-support-for-xmos-processors/3867/8)

**However** If the `CMAKE_SYSTEM_NAME` is set to `Generic` then the warning is gone. Do we need to have the system name set to something which cmake doesn't support? We could use a custom variable for differentiation until we have upstreamed support.

CiaranWoodward, 2022-08-12:
As [CMAKE_SYSTEM_NAME ](https://cmake.org/cmake/help/latest/variable/CMAKE_SYSTEM_NAME.html)is for the operating system name, putting ``set(CMAKE_SYSTEM_NAME Generic)`` in the toolchain file would be the standard thing to do for baremetal.

keithm-xmos, 2022-08-12:
We do currently use CMAKE_SYSTEM_NAME in CMakeLists to differentiate between builds targeting xcore or other (x86 usually). So, this change request has a ripple effect and will require some planning.

How would you recommend we specify that the target system is xcore? Several of our libraries can be built for x86 and xcore so we do need a flag.

CiaranWoodward, 2022-08-12:
[CMAKE_SYSTEM_PROCESSOR](https://cmake.org/cmake/help/latest/variable/CMAKE_SYSTEM_PROCESSOR.html) may be suitable because it's fairly freeform, or a custom variable like `XMOS_TARGET_PROCESSOR`.

@xhuw  is this actually breaking anything or is it just making the cmake output messy?

xhuw, 2022-08-12:
Not breaking anything, so low priority for me. Setting up a new project at the moment and raising issues on the SDK as I notice them.

The only issue is the messy output, so a fix would be a nice to have for new users, or CMake beginners who might think it's something they need to resolve

xmos-jmccarthy, 2022-08-12:
XCORE the silicon supports the XCORE operating system, so CMAKE_SYSTEM_NAME is appropriate.  Perhaps we could use CMAKE_SYSTEM_PROCESSOR for the architecture, XS3A, XS2A, etc. and XCORE for the CMAKE_SYSTEM_NAME.

I would not want to replace the current use of CMAKE_SYSTEM_NAME, as there are many useful thirdparty CMake ready libraries out there that use it explicitly for Host builds.  I wouldn't want us to be in a situation where lib_foo cmake adds something when CMAKE_SYSTEM_NAME == "Win" that I miss because I am on Linux, or vice versa.  CMAKE_SYSTEM_NAME being set to "Generic" is common in other microcontrollers, but they do not have a hardware OS like xcore does.  That being said, this is primarily just semantics.

The ideal solution would be completion of:
https://github.com/xmos/xmos_cmake_toolchain/issues/3

So that we are added to the list:
https://gitlab.kitware.com/cmake/cmake/-/blob/v3.22.0-rc2/Modules/CMakeDetermineSystem.cmake#L12-31

Then there would be no error, and we wouldn't have to submodule https://github.com/xmos/xmos_cmake_toolchain everywhere as it would be provided by the cmake installation.

keithm-xmos, 2023-03-28:
Fixed by https://github.com/xmos/xmos_cmake_toolchain/pull/4

closed 2023-03-28 by keithm-xmos