xmos/xcore_iot #508: Provide a way to validate sdk version in cmakelists

xhuw, 2022-08-16:
Thank you for submitting an SDK feature request. Please provide as much information you can.

**Describe the feature**

For my project we do not intend to distribute a copy of the SDK with the source code. This means we will be relying on customers downloading the correct version of the SDK. I want the build to fail with a sensible message if they have downloaded an unsupported version.

I could manually parse the changelog or the latest git tag. But both of these options are brittle to change.

I don't know if there is a proper cmake way to support this. Any solution that lets me query the version of the SDK at configure time is good. Or perhaps specify the required version and the SDK could fail when it notices it isn't that version.

**Will this change any current APIs? How?**

no, I think this should be an additive feature

**Who will benefit with this feature?**

me :) anyone writing a project using the SDK that wants deterministic builds but doesn't want to submodule the SDK

**Any Other info**

low priority.  If this feature request is accepted then before the end of the year would be best. If not I will parse the changelog

keithm-xmos, 2022-08-17:
Below are some ideas that pop to mind.  We will need to investigate these a bit and document a recommended solution.  

- CMake's FetchContent supports cloning (recursively if needed) the SDK repo and checking out a SHA or tag.  This happens at configure-time which could make for sad users if they need to re-configure many times.  But, if only a single configure is typical, this is a reasonable solution.  We would want to verify this works as expected.  
- The version string is in the settings.json file.  We could add a CMake macro to read and parse this. 
- It is also possible that CMake has machinery for this already.  Will need to investigate.

xhuw, 2022-08-17:
another option would be to support [find_package](https://cmake.org/cmake/help/latest/command/find_package.html) which takes a version field, then users could install multiple different versions of the sdk to their system in  a standard location. A heavy weight solution for a small problem though I expect. (I have never tried developing a library that supports this feature, so not sure what it entails)

keithm-xmos, 2023-02-17:
Wont fix.  Will reopen in future if necessary.

closed 2023-02-17 by keithm-xmos