xmos/xcore_iot #512: Request support for pll control in SDK

xhuw, 2022-08-31:
Thank you for submitting an SDK feature request. Please provide as much information you can.

**Describe the feature**

app_pll_ctrl.c is being copy pasta'd into all examples and projects (9 places so far):
https://github.com/search?q=org%3Axmos++extension%3Ac++filename%3Aapp_pll_ctrl&type=Code&ref=advsearch&l=&l=

Can this be supported by the SDK instead?

**Will this change any current APIs? How?**

don't think so

**Who will benefit with this feature?**

any project initialising or using the app pll

**Any Other info**

Thanks :D

ed-xmos, 2022-09-01:
Note that @XMOS-JoeG has code to monotonically control the PLL rate with quite good linearity.

XMOS-JoeG, 2022-09-01:
I think the code to initialise the app PLL should be part of the tools as it needs to work around a chip bug and we may need to update that code if we find it doesn't work in all scenarios. Also the nature of the bug is pretty complex so shouldn't really be left to the user to work out/debug (they would have no chance without understanding the detailed architecutre).
See my suggestion for a helper function here:
http://bugzilla/show_bug.cgi?id=18380

xmos-jmccarthy, 2022-09-26:
This issue should be resolved by adding the app pll functions to:
https://github.com/xmos/fwk_core/tree/develop/modules/clock_control
Until the tools natively support it.