xmos/xcore_iot #534: Cloning fails with: git@github.com: Permission denied (publickey)

richardklingler, 2022-10-12:
**System information**
 - macOS 12.6

**Problem**

As per official documentation the sdk should be checked out with:

   git clone --recurse-submodules https://github.com/xmos/xcore_sdk.git

Which fails when cloning the sub-modules:

Cloning into 'xcore_sdk'...
remote: Enumerating objects: 25733, done.
remote: Counting objects: 100% (424/424), done.
remote: Compressing objects: 100% (224/224), done.
remote: Total 25733 (delta 208), reused 386 (delta 195), pack-reused 25309
Receiving objects: 100% (25733/25733), 47.76 MiB | 15.49 MiB/s, done.
Resolving deltas: 100% (15687/15687), done.
Submodule 'frameworks/core' (git@github.com:xmos/fwk_core.git) registered for path 'modules/core'
Submodule 'frameworks/io' (git@github.com:xmos/fwk_io.git) registered for path 'modules/io'
Submodule 'frameworks/rtos' (git@github.com:xmos/fwk_rtos.git) registered for path 'modules/rtos'
Submodule 'lib_src' (git@github.com:xmos/lib_src.git) registered for path 'modules/sample_rate_conversion/lib_src'
Submodule 'avona' (git@github.com:xmos/sw_avona.git) registered for path 'modules/voice'
Submodule 'xmos_cmake_toolchain' (git@github.com:xmos/xmos_cmake_toolchain.git) registered for path 'xmos_cmake_toolchain'
Cloning into '/Users/klingler/Develop/xmos/xcore_sdk/modules/core'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:xmos/fwk_core.git' into submodule path '/Users/klingler/Develop/xmos/xcore_sdk/modules/core' failed
Failed to clone 'modules/core'. Retry scheduled
Cloning into '/Users/klingler/Develop/xmos/xcore_sdk/modules/io'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:xmos/fwk_io.git' into submodule path '/Users/klingler/Develop/xmos/xcore_sdk/modules/io' failed
Failed to clone 'modules/io'. Retry scheduled
Cloning into '/Users/klingler/Develop/xmos/xcore_sdk/modules/rtos'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:xmos/fwk_rtos.git' into submodule path '/Users/klingler/Develop/xmos/xcore_sdk/modules/rtos' failed
Failed to clone 'modules/rtos'. Retry scheduled
Cloning into '/Users/klingler/Develop/xmos/xcore_sdk/modules/sample_rate_conversion/lib_src'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:xmos/lib_src.git' into submodule path '/Users/klingler/Develop/xmos/xcore_sdk/modules/sample_rate_conversion/lib_src' failed
Failed to clone 'modules/sample_rate_conversion/lib_src'. Retry scheduled
Cloning into '/Users/klingler/Develop/xmos/xcore_sdk/modules/voice'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:xmos/sw_avona.git' into submodule path '/Users/klingler/Develop/xmos/xcore_sdk/modules/voice' failed
Failed to clone 'modules/voice'. Retry scheduled
Cloning into '/Users/klingler/Develop/xmos/xcore_sdk/xmos_cmake_toolchain'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:xmos/xmos_cmake_toolchain.git' into submodule path '/Users/klingler/Develop/xmos/xcore_sdk/xmos_cmake_toolchain' failed
Failed to clone 'xmos_cmake_toolchain'. Retry scheduled
Cloning into '/Users/klingler/Develop/xmos/xcore_sdk/modules/core'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:xmos/fwk_core.git' into submodule path '/Users/klingler/Develop/xmos/xcore_sdk/modules/core' failed
Failed to clone 'modules/core' a second time, aborting

keithm-xmos, 2022-10-12:
Hi @richardklingler, I apologize that you have been unable to recursively clone the repository.  The repo currently requires that you have SSH authentication setup for github.com.  We expect to resolve this by the first release, however, it is required for the pre-releases.  See https://docs.github.com/en/authentication/connecting-to-github-with-ssh for more info on how to set up github.com SSH authentication.

richardklingler, 2022-10-12:
Hello Keith

Thanks for the quick reply...thought it was something like that (o;

Just ordered a XK-EVK-XE216 as it seems to be the only board left to start developing...all other boards are obsolete @digikey....but as this repo seems to be still updated I hope xmos products will be as well (o;

richardklingler, 2022-10-12:
Hmm...there is no SDK for the XK-EVK-XE216 as it is obsolete now and isn't supported anymore?

BTW: No search results on xmos.ai when searching XK-EVK-XU316 (o;

keithm-xmos, 2022-10-12:
Support for XCORE-200 boards is no longer officially supported.

richardklingler, 2022-10-12:
Means I can't use it anymore and Digikey still charges full price for something that I can throw to trash directly?

keithm-xmos, 2022-10-12:
Please allow me to clarify.  The example applications in this repository are only supported on the XK-EVK-XU316 board.  XCORE-200 kits are still supported in general.  To use your XK-EVK-XE216, I recommend you start here:  https://www.xmos.ai/libraries/

richardklingler, 2022-10-16:
Right...thanks for clarification...

Sadly the XK-EVK-XU316 kit isn't available anywhere as those AI kits for speech are not interesting...

closed 2022-10-22 by keithm-xmos