xmos/xcore_iot #570: Installation Step 2. Install Host Applications Fails on Mac M1

VoyageAudio, 2022-12-19:
**System information**
 - OS: OSX 12.6 Monterey Running on MacBook Pro (2021, M1 Pro)

**Describe the problem**
XCORE Installation instructions Step # 2 fails on M1 arch.

**Provide the exact sequence of commands / steps that you executed before running into the problem**
Please see attached image.

[image]

keithm-xmos, 2022-12-20:
Hi @VoyageAudio, I apologize that you encountered an issue building on your M1.  I have a couple work-arounds for you to try.  

First, you can force CMake to build the x86 host app by adding the following line to the top of the file `examples/freertos/device_control/host/CMakeLists.txt`:

`set(CMAKE_OSX_ARCHITECTURES "x86_64" CACHE INTERNAL "")`

The x86 host app will run on an M1.  Alternatively, you can bypass the device control host app build by commenting out this line in `examples/examples.cmake`:

`add_subdirectory(${CMAKE_CURRENT_LIST_DIR}/freertos/device_control/host)`

It looks like it is possible to build for the M1.  I have added a new issue to investigate and fix this rather than provide a work-around.  https://github.com/xmos/xcore_sdk/issues/571

VoyageAudio, 2022-12-20:
Thank you for this. I think it gets past the device control host app building using this work around, however it fails immediately after because it can't find "xscope_endpoint.h". This is completely unrelated, correct?

`~GitHub/xcore_sdk/examples/freertos/tracealyzer/host/xscope2psf.c:9:10: fatal error: 'xscope_endpoint.h' file not found
#include "xscope_endpoint.h"
               ^~~~~~~~~~~~~~~~~~~
         
1 error generated.
make[2]: *** [examples/freertos/tracealyzer/host/CMakeFiles/xscope2psf.dir/xscope2psf.c.o] Error 1
make[1]: *** [examples/freertos/tracealyzer/host/CMakeFiles/xscope2psf.dir/all] Error 2
make: *** [all] Error 2
bash-3.2$ /xcore_sdk/examples/freertos/tracealyzer/host/xscope2psf.c:9:10: fatal error: 'xscope_endpoint.h' file not found
bash: /xcore_sdk/examples/freertos/tracealyzer/host/xscope2psf.c:9:10:: No such file or directory
`

keithm-xmos, 2022-12-20:
Yes, this is unrelated but easy to fix.

You need to ensure that your XTC Tools environment is setup when building.  This is true even for the host applications.  The official documentation for configuring your command-line environment is here: https://www.xmos.ai/documentation/XM-014363-PC-6/html/tools-guide/install-configure/getting-started.html

Another, equally valid way is to run:

`source /Applications/XMOS_XTC_15.1.4/SetEnv.sh`

because sourcing SetEnv.sh is what the SetEnv.command does.

VoyageAudio, 2022-12-20:
Yeah, I've done all of that. Tried both ways, same error as result.

keithm-xmos, 2022-12-20:
Ensure you have these lines in your examples/freertos/tracealyzer/host/CMakeLists.txt: 

https://github.com/xmos/xcore_sdk/blob/main/examples/freertos/tracealyzer/host/CMakeLists.txt#L25-L27

If you do then I am not sure why the include folder is not set.  Adding @jcarrier-xmos in case he has any suggestions.

VoyageAudio, 2022-12-20:
Thanks. I'm continuing this thread because its all related to building on Apple Silicon. I changed the line you pointed out to the absolute path as a work around and I seem to get past this issue.

Next up there is something else wrong:
[image]

It looks like for some reason it's adding an additional ".0".

keithm-xmos, 2022-12-23:
It is not clear from this message which Executable CMake is unable to find.  One idea that may be helpful is to re-run the CMake configure step and generate an output of all commands.  Run the following in the `build_host` directory:

```
cmake .. -DCMAKE_EXPORT_COMPILE_COMMANDS=1
```

Then share the `compile_commands.json` file here and we will investigate.  

In the meantime, you can likely bypass the host application builds by simply commenting any of the lines in `examples/examples.cmake`.  See https://github.com/xmos/xcore_sdk/blob/main/examples/examples.cmake#L29-L32

VoyageAudio, 2022-12-23:
Thanks. I cloned a fresh xcore_sdk and am getting slightly different results now - perhaps there have been some recent changes?

Here is the output now without any modifications above:
[image]

I attached the .json file you requested:
[compile_commands.json.zip](https://github.com/xmos/xcore_sdk/files/10296527/compile_commands.json.zip)

Going to bypass his applications as you mention - thanks for that!

keithm-xmos, 2022-12-23:
This compilation error has been fixed on the latest develop branch.  

Thanks for the compile_commands.json, I will review it.