xmos/xcore_iot #578: AI explorer board documentation

AndrewCapon, 2023-01-22:
Hi Guys,

Sorry, I know this is probably the wrong place to post this but I have had no reply from support for nearly a month.

The hardware manual at https://www.xmos.ai/download/xcore.ai-explorer-board-v2.0-hardware-manual(5).pdf is out of date and the links do not work.

I have found the data sheets but the following application notes I cannot find:

XM-014230-AN
XM-014200-AN
XM-014251-PC
XM-014203-AN

Are these available anywhere, or an updated hardware manual?

Many thanks

Andy

keithm-xmos, 2023-01-23:
Hi @AndrewCapon, I apologize that you have not heard back from support.  I have reached out to them and we are working on getting you these documents.

keithm-xmos, 2023-01-23:
I have attached the app note documents.  

[supplemental-docs-unapproved.zip](https://github.com/xmos/xcore_sdk/files/10480910/supplemental-docs-unapproved.zip)

AndrewCapon, 2023-01-23:
Hi @keithm-xmos,

Thanks very much for your help with this.

Cheers

Andy

AndrewCapon, 2023-01-23:
Hi @keithm-xmos,

Do you have any info on the wifi side of things, the doc is empty.

Just some example code would do.

Many Thanks

Andy

keithm-xmos, 2023-01-23:
Our support team has created a ticket to update this document.  I apologize once again.

AndrewCapon, 2023-01-24:
Thanks @keithm-xmos

AndrewCapon, 2023-04-03:
> Our support team has created a ticket to update this document. I apologize once again.

AndrewCapon, 2023-04-03:
The hardware manual is here: https://www.xmos.ai/download/xcore.ai-explorer-board-v2.0-hardware-manual(5).pdf

It is still hopelessly out of date, the hardware is actually wrong.

Most of the links do not work, the AP Notes are not there, they are not correct in the zip you linked to.

Nothing has changed.

Do you think this is acceptable?