xmos/xcore_iot #584: CMake error during installation

PelleS0205, 2023-02-22:
**System information**
 - OS: Ubuntu 20.04.5 
 - SDK version: 0.12.0

**Problem**
When installing the SDK, CMake gives the following error and fails:

_CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
XSCOPE_ENDPOINT_LIB
    linked by target "xscope2psf" in directory /xcore_sdk/examples/freertos/tracealyzer/host
    linked by target "xscope_host_endpoint" in directory /xcore_sdk/modules/xscope_fileio/xscope_
_

**Steps**
1. git clone --recurse-submodules https://github.com/xmos/xcore_sdk.git
2. cmake -B build_host

keithm-xmos, 2023-02-23:
Hi @PelleS0205,

I apologize you ran into this issue building the host applications.  Some host applications require that the location of `xscope_endpoint.so` be added to your `LD_LIBRARY_PATH` environment variable.  This environment variable will be set if you run the host application build in the XTC Tools command-line environment.  

Or, you may prefer to set this environment variable manually.

        export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:<path-to-XTC-Tools>/lib

This should resolve your issue.

keithm-xmos, 2023-02-24:
Do you get an error when you run?

    sudo make install

closed 2023-03-07 by keithm-xmos