xmos/xcore_iot #586: Can't clone the submodules within the repository

NitinMedisetti, 2023-03-03:
"Cloning into 'C:/Users/{...}/xcore_sdk/modules/core'...
Host key verification failed.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:xmos/{......}' into submodule path 'C:/Users/{...}/xcore_sdk/modules/core' failed
Failed to clone 'modules/core'. Retry scheduled"

This showed up for all the submodules.

keithm-xmos, 2023-03-07:
Hi @NitinMedisetti, I apologize that you ran into this issue.   In order to clone all the submodules, you need to have SSH authentication setup on GitHub.com.  See https://docs.github.com/en/authentication/connecting-to-github-with-ssh

NitinMedisetti, 2023-03-08:
Hi @keithm-xmos, I've set up SSH authentication with the help of the link you've provided. Even tested it, as shown in https://docs.github.com/en/authentication/connecting-to-github-with-ssh/testing-your-ssh-connection
I still can't clone the submodules. It's same as before.

keithm-xmos, 2023-03-08:
Let's try to clone one of the submodules.  Does this command work for you?

```
git clone git@github.com:xmos/lib_xcore_math.git
```

If the above command does work, try these steps to clone the xcore_sdk:

```
git clone git@github.com:xmos/xcore_sdk.git
cd xcore_sdk
git submodule update --init --recursive
```

Please let me know how this goes.  Thanks @keithm-xmos

NitinMedisetti, 2023-03-08:
This is what happened.

[image]

This created empty folders for submodels. Just like before.

NitinMedisetti, 2023-03-09:
Oh, my bad. 
Okay, I've added "update". Now I'm back to where I've started.
[image]

NitinMedisetti, 2023-03-09:
Now, I've manually downloaded each submodule. But I just wanted to know what caused the issue and how to deal with it.

keithm-xmos, 2023-03-09:
Can you please clarify what commands you ran to download each submodule?  Many of the submodules also contain submodules.  It would be interesting if those work but the xcore_sdk does not.  For example, does this series of commands work for you?

```
git clone git@github.com:xmos/fwk_core.git
cd fwk_core
git submodule update --init --recursive
```

Thanks again!

NitinMedisetti, 2023-03-09:
I didn't use any commands to download submodules. I've manually navigated to each of those submodules in GitHub and downloaded them as zip files.
```
git clone git@github.com:xmos/fwk_core.git
cd fwk_core
git submodule update --init --recursive
```
And these commands didn't work. It's the same as before. 
"Please make sure you have the correct access rights and the repository exists"

keithm-xmos, 2023-03-09:
OK.  Your zip downloads will only work if you walked through each repo and also downloaded its submodules.  If you missed any, you may run into issues trying to build the applications.  

Does this work for you?

```
git clone git@github.com:xmos/lib_xcore_math.git
```

NitinMedisetti, 2023-03-09:
`git clone git@github.com:xmos/lib_xcore_math.git`
Yes this works

And, I made sure I didn't miss any submodules, and the application was built successfully.

keithm-xmos, 2023-03-14:
Closing issue as we are unable to reproduce.

closed 2023-03-14 by keithm-xmos