xmos/xcore_iot #591: Merge the branch used in xvf3800 to develop

shuchitak, 2023-03-20:
XVF3800 currently uses the branch https://github.com/xmos/xcore_sdk/tree/feature/xvf3800_testing which was branched off from the commit 008b93a43081301e5363fea8b0f025a4feb85451 and has fixes we've made since then as part of sw_xvf3800 development. 
This branch needs to be merged to xcore_sdk develop and sw_xvf3800 needs to be updated to use the latest xcore_sdk.

keithm-xmos, 2023-04-03:
Can this issue be closed now that sw_xvf3800 no longer depends on the the xcore_sdk repository?

keithm-xmos, 2023-04-12:
I should have tagged @shuchitak.

Can this issue be closed now that sw_xvf3800 no longer depends on the the xcore_sdk repository?

shuchitak, 2023-04-17:
@keithm-xmos, yes, I'll close this.

closed 2023-04-17 by shuchitak