xmos/xcore_iot #627: Getting started instructions produce Makefile:32: *** missing separator.  Stop.

snellejelle99, 2023-07-21:
**System information**
 - OS: Windows 10
 - xcore board NA
 - Version tag or commit SHA 1.0.0

**Describe the current behavior**
When executing the following commands from the [getting started docs](https://www.xmos.ai/documentation/XM-014660-PC-2/html/doc/quick_start/installation.html#sdk-installation):

> cmake  -G "NMake Makefiles" -B build_host
> cd build_host
> make install

I get the following error:
> Makefile:32: *** missing separator.  Stop.

**Describe the expected behavior**
I expect this error to not happen.

**Standalone code to reproduce the issue**
Provided above

**Other info or logs** 
Makefile provided (.txt was added to satisfy githubs file requirements)
[Makefile.txt](https://github.com/xmos/xcore_iot/files/12122876/Makefile.txt)
cmake cache file provided
[CMakeCache.txt](https://github.com/xmos/xcore_iot/files/12122872/CMakeCache.txt)

uvvpavel, 2023-08-08:
Hi, Thanks for spotting an error in our documentation. It should say to use `nmake` rather than `make`. If you don't have `nmake`  you can also try `ninja`  (which will be slightly faster) but you'll need to do this instead:
```
cmake -G Ninja -B build_host
cd build_host
ninja
```
I've tried it with `nmake` and `ninja` and it worked fine. Let me know if it fixes your build ✨

snellejelle99, 2023-08-28:
Yes, this seems to have fixed the problem.

closed 2023-08-28 by snellejelle99