xmos/xcore_iot #631: Readme update - git clone --recurse-submodules fails

netkruzer, 2024-06-25:
The Readme file says to 
`git clone --recurse-submodules git@github.com:xmos/xcore_iot.git`
but it fails on many submodules.  I'm logged in with my Github SSH key. It retrieves some, but not all.

This worked for me:
`git clone --recurse-submodules https://github.com/xmos/xcore_iot.git`

I'm using Ubuntu 22.04.4. Also had the same issue on Windows in git Bash.

This is my error log:

```
kevin@NUC-TV:~/Downloads/XMOS$ git clone --recurse-submodules git@github.com:xmos/xcore_iot.git
Cloning into 'xcore_iot'...
remote: Enumerating objects: 27436, done.
remote: Counting objects: 100% (2142/2142), done.
remote: Compressing objects: 100% (968/968), done.
remote: Total 27436 (delta 1281), reused 1839 (delta 1130), pack-reused 25294
Receiving objects: 100% (27436/27436), 48.26 MiB | 14.45 MiB/s, done.
Resolving deltas: 100% (16749/16749), done.
Submodule 'frameworks/core' (git@github.com:xmos/fwk_core.git) registered for path 'modules/core'
Submodule 'modules/inferencing/lib_nn' (git@github.com:xmos/lib_nn.git) registered for path 'modules/inferencing/lib_nn'Submodule 'modules/inferencing/lib_tflite_micro' (git@github.com:xmos/lib_tflite_micro.git) registered for path 'modules/inferencing/lib_tflite_micro'
Submodule 'frameworks/io' (git@github.com:xmos/fwk_io.git) registered for path 'modules/io'
Submodule 'lib_qspi_fast_read' (git@github.com:xmos/lib_qspi_fast_read.git) registered for path 'modules/qspi_fast_read'Submodule 'frameworks/rtos' (git@github.com:xmos/fwk_rtos.git) registered for path 'modules/rtos'
Submodule 'lib_src' (git@github.com:xmos/lib_src.git) registered for path 'modules/sample_rate_conversion/lib_src'
Submodule 'modules/xscope_fileio/xscope_fileio' (git@github.com:xmos/xscope_fileio.git) registered for path 'modules/xscope_fileio/xscope_fileio'
Submodule 'xmos_cmake_toolchain' (git@github.com:xmos/xmos_cmake_toolchain.git) registered for path 'xmos_cmake_toolchain'
Cloning into '/home/kevin/Downloads/XMOS/xcore_iot/modules/core'...
remote: Enumerating objects: 558, done.
remote: Counting objects: 100% (83/83), done.
remote: Compressing objects: 100% (55/55), done.
remote: Total 558 (delta 26), reused 60 (delta 22), pack-reused 475
Receiving objects: 100% (558/558), 144.44 KiB | 1.16 MiB/s, done.
Resolving deltas: 100% (233/233), done.
Cloning into '/home/kevin/Downloads/XMOS/xcore_iot/modules/inferencing/lib_nn'...
remote: Enumerating objects: 12582, done.
remote: Counting objects: 100% (1268/1268), done.
remote: Compressing objects: 100% (499/499), done.
remote: Total 12582 (delta 884), reused 1019 (delta 768), pack-reused 11314
Receiving objects: 100% (12582/12582), 3.50 MiB | 8.56 MiB/s, done.
Resolving deltas: 100% (8755/8755), done.
Cloning into '/home/kevin/Downloads/XMOS/xcore_iot/modules/inferencing/lib_tflite_micro'...
remote: Enumerating objects: 4999, done.
remote: Counting objects: 100% (2313/2313), done.
remote: Compressing objects: 100% (955/955), done.
remote: Total 4999 (delta 1457), reused 1564 (delta 1337), pack-reused 2686
Receiving objects: 100% (4999/4999), 14.91 MiB | 12.81 MiB/s, done.
Resolving deltas: 100% (3065/3065), done.
Cloning into '/home/kevin/Downloads/XMOS/xcore_iot/modules/io'...
remote: Enumerating objects: 2870, done.
remote: Counting objects: 100% (645/645), done.
remote: Compressing objects: 100% (288/288), done.
remote: Total 2870 (delta 387), reused 486 (delta 338), pack-reused 2225
Receiving objects: 100% (2870/2870), 2.10 MiB | 1.20 MiB/s, done.
Resolving deltas: 100% (1442/1442), done.
Cloning into '/home/kevin/Downloads/XMOS/xcore_iot/modules/qspi_fast_read'...
remote: Enumerating objects: 563, done.
remote: Counting objects: 100% (216/216), done.
remote: Compressing objects: 100% (110/110), done.
remote: Total 563 (delta 123), reused 176 (delta 104), pack-reused 347
Receiving objects: 100% (563/563), 129.91 KiB | 1.06 MiB/s, done.
Resolving deltas: 100% (282/282), done.
Cloning into '/home/kevin/Downloads/XMOS/xcore_iot/modules/rtos'...
remote: Enumerating objects: 28430, done.
remote: Counting objects: 100% (2038/2038), done.
remote: Compressing objects: 100% (859/859), done.
remote: Total 28430 (delta 1207), reused 1775 (delta 1127), pack-reused 26392
Receiving objects: 100% (28430/28430), 59.28 MiB | 13.66 MiB/s, done.
Resolving deltas: 100% (17305/17305), done.
Cloning into '/home/kevin/Downloads/XMOS/xcore_iot/modules/sample_rate_conversion/lib_src'...
remote: Enumerating objects: 5527, done.
remote: Counting objects: 100% (528/528), done.
remote: Compressing objects: 100% (134/134), done.
remote: Total 5527 (delta 411), reused 412 (delta 394), pack-reused 4999
Receiving objects: 100% (5527/5527), 4.99 MiB | 10.87 MiB/s, done.
Resolving deltas: 100% (3352/3352), done.
Cloning into '/home/kevin/Downloads/XMOS/xcore_iot/modules/xscope_fileio/xscope_fileio'...
ssh: connect to host github.com port 22: Connection timed out
**fatal: Could not read from remote repository**.

**Please make sure you have the correct access rights**
and the repository exists.
fatal: clone of 'git@github.com:xmos/xscope_fileio.git' into submodule path '/home/kevin/Downloads/XMOS/xcore_iot/modules/xscope_fileio/xscope_fileio' failed
Failed to clone 'modules/xscope_fileio/xscope_fileio'. Retry scheduled
Cloning into '/home/kevin/Downloads/XMOS/xcore_iot/xmos_cmake_toolchain'...
ssh: connect to host github.com port 22: Connection timed out
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:xmos/xmos_cmake_toolchain.git' into submodule path '/home/kevin/Downloads/XMOS/xcore_iot/xmos_cmake_toolchain' failed
Failed to clone 'xmos_cmake_toolchain'. Retry scheduled
Cloning into '/home/kevin/Downloads/XMOS/xcore_iot/modules/xscope_fileio/xscope_fileio'...
ssh: connect to host github.com port 22: Connection timed out
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:xmos/xscope_fileio.git' into submodule path '/home/kevin/Downloads/XMOS/xcore_iot/modules/xscope_fileio/xscope_fileio' failed
Failed to clone 'modules/xscope_fileio/xscope_fileio' a second time, aborting
```

netkruzer, 2024-06-25:
The issue is also on this documentation page:

https://www.xmos.com/documentation/XM-014926-PC-1/html/doc/programming_guide/installation.html

ACascarino, 2024-07-02:
Hi @netkruzer,

Unfortunately, attempts to reproduce this issue (including testing with users not part of XMOS) have failed, with all attempts to clone this repository and its submodules succeeding - including xscope_fileio.

I apologise for the lack of resolution on this, but I'm afraid that without the ability to reproduce locally I'm unable to offer further advice. 

If you do find the root cause of the issue however, please do feel free to add it here - other people may benefit from it in the future.

Kind regards,
Angel Cascarino,
Software Engineer, XMOS

closed 2024-07-02 by ACascarino

netkruzer, 2024-07-02:
I was able to find the binary zip file for sw_usb_audio, and that worked for me instead of having to recursively clone all the repos. 

https://github.com/xmos/sw_usb_audio/releases/tag/v8.1.0

Is there a way to zip everything together for the xcore_iot in it's release folder?

netkruzer, 2024-07-03:
I found this post on [StackOverflow](https://stackoverflow.com/questions/15589682/how-to-fix-ssh-connect-to-host-github-com-port-22-connection-timed-out-for-g):

Create a `config` file in the `.ssh` folder:

```
Host github.com
User git
Hostname ssh.github.com
PreferredAuthentications publickey
IdentityFile ~/.ssh/id_rsa
Port 443
```